robocasa/robocasa #114: mosaic when rendering videos

StarCycle, 2024-12-19:
Hello @snasiriany @fangbohui @yukezhu @Steve-Tod,

I installed robocasa and render the dataset with the command `python -m robocasa.demos.demo_tasks --render_offscreen`. However, I get videos with mosaic. These are the renderred videos:

https://github.com/user-attachments/assets/356f00f2-22c0-4ebd-a89d-6bac26f6d4e1

https://github.com/user-attachments/assets/dc365b25-09f5-490e-9aa2-319cf413fa3f

https://github.com/user-attachments/assets/4389cb57-e729-4cba-97a6-d85d1789d2f5

My GPU is A100 80g. Do you have any clue about it?

snasiriany, 2024-12-19:
Hi @StarCycle , I tried this on two linux workstations (one with Titan RTX the other with RTX A5000). I could not reproduce this issue.

I'm using the latest commit id's from robosuite and robocasa:
robosuite: d340b677d87d37c26aace2cc842e2a0f4d8f5fa7
robocasa: 00b69d8bf7ae41bceb511c3235784d696c1e867a

Furthermore, I'm using MuJoCo 3.2.6.

What are your commit id's, mujoco version, and cuda version?

One thing to check is to see which type of rendering is being used in robosuite:
https://github.com/ARISE-Initiative/robosuite/blob/d340b677d87d37c26aace2cc842e2a0f4d8f5fa7/robosuite/utils/binding_utils.py#L68-L73

There is CPU rendering (osmesa) and GPU rendering (egl). Can you put some prints/breakpoints here to find that out and share with that with us?

snasiriany, 2024-12-22:
Feel free to reply if you have any updates

closed 2024-12-22 by snasiriany

StarCycle, 2024-12-23:
@snasiriany Thanks but let me transfer to another benchmark for a while...

alek98, 2025-06-02:
I'm experiencing the same issue.

**Setup:**

* MuJoCo 3.2.6
* CUDA 12.6
* Commit ID: `0eae0634a61ad2be33962c9de7000a2dd1ee573f` (latest)
* Running on A100
* Using EGL

Note:
I had to upgrade PyOpenGL from the required 3.1.0 version (I tried 3.1.4 and 3.1.9) due to pointer errors encountered when using EGL.

Here's minimal code to reproduce the issue:

```python
import os
os.environ['MUJOCO_GL'] = 'egl'

from robocasa.environments import ALL_KITCHEN_ENVIRONMENTS
from robocasa.utils.env_utils import create_env, run_random_rollouts
import numpy as np

# Choose a random task
env_name = np.random.choice(list(ALL_KITCHEN_ENVIRONMENTS))

env = create_env(
    env_name=env_name,
    render_onscreen=False,
    camera_widths=256,
    camera_heights=256,
    seed=0,  # Set seed=None to run unseeded
)

# Reset the environment (uncomment if needed)
env.reset()

imgs = []
for i in range(50):
    action = np.random.randn(*env.action_spec[0].shape) * 0.1
    obs, reward, done, info = env.step(action)
    img = np.flipud(obs['robot0_agentview_right_image'])
    imgs.append(img)
```

Now `imgs` contain images with rendering issue. Note that the same happens when using `env.sim.render()`.

Please let me know if you can reproduce this issue on your end.