robocasa/robocasa #115: Pretrained model weight in RoboCasa paper

andvg3, 2024-12-31:
Hi,

Thank you for your great contribution to the E-AI community. I'm interested in adapting your model for finetuning in new environments. Do you provide the pretrained weights for the model in your paper? If so, could you kindly provide me some instructions on how to obtain them?

Thanks in advance! Best, An

snasiriany, 2025-01-02:
Hi @andvg3, our model is trained on a somewhat older (and private) version of the codebase. We are also planning to release some new models within a couple months, so we will hold off on releasing model weights until the next project update.

You can reproduce the same results, training the same model and same datasets by following the instructions here: https://robocasa.ai/docs/use_cases/policy_learning.html

Hope this helps!

closed 2025-01-02 by snasiriany

andvg3, 2025-01-08:
> Hi @andvg3, our model is trained on a somewhat older (and private) version of the codebase. We are also planning to release some new models within a couple months, so we will hold off on releasing [...]

Hello, I'm having trouble loading the MG-3000 dataset onto my A100 40GB GPU. Could you share the computer specifications you used for training? Additionally, what are possible ways to reduce the memory load? Thank you!

snasiriany, 2025-01-08:
Your A100 40Gb GPU should definitely be sufficient. We used a A5000 GPU with 24Gb. Can you share a stack trace, and additional details about your setup? Are you using the robomimic branch that we provided for policy training?