robocasa/robocasa #120: Running basic usage code on MacOS

haroonmoh, 2025-01-20:
When I run the basic usage code on MacOS (Sequoia 15.1), I get the error below:

*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow should only be instantiated on the main thread!'
*** First throw call stack:
(
        0   CoreFoundation                      0x00007ff81a6f86e6 __exceptionPreprocess + 242
        1   libobjc.A.dylib                     0x00007ff81a1e0bd0 objc_exception_throw + 62
        2   CoreFoundation                      0x00007ff81a71b654 -[NSException raise] + 9
        3   AppKit                              0x00007ff81dfb5150 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1653
        4   AppKit                              0x00007ff81dfb4ad2 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
        5   libglfw.3.dylib                     0x000000010fd3ae7b _glfwPlatformCreateWindow + 411
        6   libglfw.3.dylib                     0x000000010fd342ca glfwCreateWindow + 474
        7   libffi.dylib                        0x00007ff82cac9882 ffi_call_unix64 + 82
        8   ???                                 0x0000700001b76770 0x0 + 123145331107696
)
libc++abi: terminating due to uncaught exception of type NSException

Seems to be a similar issue as one posted on the Mujoco Github. However, I wasn't able to find the line causing the error.

snasiriany, 2025-01-21:
Interesting, did this happen for both code samples in getting started page?

Online rollouts: https://robocasa.ai/docs/use_cases/basic_usage.html#gym-interface
Offline rollouts: https://robocasa.ai/docs/use_cases/basic_usage.html#offscreen-rollouts

To ensure that this isn't a robocasa-specific issue, can you try the getting started example in robosuite: https://robosuite.ai/docs/basicusage.html#basic-usage

slolla, 2025-01-24:
Had the same issue-- the robosuite getting started example works fine, but both online and offline rollouts in the basic robocasa usage result in the same error mentioned above.

snasiriany, 2025-01-24:
I'm wondering if this is a core MuJoCo issue. Can you please try to download this asset: https://github.com/google-deepmind/mujoco/blob/main/doc/_static/example.xml

And run `python -m mujoco.viewer --mjcf example.xml`?

Also, please try one of the asset examples that we already have:
```
python -m mujoco.viewer --mjcf robocasa/models/assets/fixtures/stoves/basic_sleek_induc/model.xml
```

If the same issue is present with these example, there is an issue with your mujoco installation.

slolla, 2025-01-24:
Both examples you just linked work for me! I think the issue is with running the code using `mjpython`: when I use `python -m` to run the basic usage code, I have no errors, but the tutorial (and mujoco website) state that we have to use `mjpython`?

haroonmoh, 2025-01-25:
Sorry for the late response but the Robosuite basic usage code and Mujoco assets also work for me. Also as @slolla, the `python -m` works for the basic usage code, but fails when I try `mjpython`.

snasiriany, 2025-01-26:
Interesting that the issue doesn't happen for the Robosuite basic usage code, but it does happen for RoboCasa.

Can you please try to run this more complete script from the Robosuite side to see if you have the issue?
```
mjpython -m robosuite.scripts.collect_human_demonstrations --environment Lift --device keyboard
```

And also a similar script with RoboCasa:
```
mjpython -m robocasa.scripts.collect_demos --environment PnPCounterToCab --device keyboard
```

haroonmoh, 2025-01-29:
Both of them work. I'm not sure why it is isolated to the basic usage code?

snasiriany, 2025-02-02:
I did a big more digging and I was able to reproduce this `NSInternalInconsistencyException`, in my case I get this:
```
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow should only be instantiated on the main thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x0000000194ce22ec __exceptionPreprocess + 176
	1   libobjc.A.dylib                     0x00000001947c6158 objc_exception_throw + 60
	2   CoreFoundation                      0x0000000194d07268 _CFBundleGetValueForInfoKey + 0
	3   AppKit                              0x00000001984dc2a8 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 188
	4   AppKit                              0x00000001984dc1e0 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 48
	5   libglfw.3.dylib                     0x0000000100e0d710 _glfwPlatformCreateWindow + 336
	6   libglfw.3.dylib                     0x0000000100e07bc8 glfwCreateWindow + 332
	7   libffi.8.dylib                      0x000000010077804c ffi_call_SYSV + 76
	8   libffi.8.dylib                      0x0000000100775834 ffi_call_int + 1404
	9   _ctypes.cpython-310-darwin.so       0x000000010075811c _ctypes_callproc + 944
	10  _ctypes.cpython-310-darwin.so       0x00000001007523f8 PyCFuncPtr_call + 228
	11  python3.10                          0x0000000100a8b664 _PyEval_EvalFrameDefault + 59140
	12  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	13  python3.10                          0x0000000100aa5d94 call_function + 148
	14  python3.10                          0x0000000100a7ec38 _PyEval_EvalFrameDefault + 7384
	15  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	16  python3.10                          0x00000001009744a4 method_vectorcall + 124
	17  python3.10                          0x0000000100a86c8c _PyEval_EvalFrameDefault + 40236
	18  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	19  python3.10                          0x00000001009fbb9c slot_tp_init + 540
	20  python3.10                          0x00000001009f10b8 type_call + 284
	21  python3.10                          0x0000000100aa5f90 call_function + 656
	22  python3.10                          0x0000000100a7ec38 _PyEval_EvalFrameDefault + 7384
	23  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	24  python3.10                          0x00000001009744a4 method_vectorcall + 124
	25  python3.10                          0x0000000100aa5d94 call_function + 148
	26  python3.10                          0x0000000100a7ec38 _PyEval_EvalFrameDefault + 7384
	27  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	28  python3.10                          0x00000001009fbb9c slot_tp_init + 540
	29  python3.10                          0x00000001009f10b8 type_call + 284
	30  python3.10                          0x0000000100aa5f90 call_function + 656
	31  python3.10                          0x0000000100a7ec38 _PyEval_EvalFrameDefault + 7384
	32  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	33  python3.10                          0x00000001009744a4 method_vectorcall + 124
	34  python3.10                          0x0000000100a86c8c _PyEval_EvalFrameDefault + 40236
	35  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	36  python3.10                          0x00000001009744a4 method_vectorcall + 124
	37  python3.10                          0x0000000100a86c8c _PyEval_EvalFrameDefault + 40236
	38  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	39  python3.10                          0x00000001009744a4 method_vectorcall + 124
	40  python3.10                          0x0000000100a86c8c _PyEval_EvalFrameDefault + 40236
	41  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	42  python3.10                          0x0000000100a8547c _PyEval_EvalFrameDefault + 34076
	43  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	44  python3.10                          0x00000001009744a4 method_vectorcall + 124
	45  python3.10                          0x0000000100aa5d94 call_function + 148
	46  python3.10                          0x0000000100a7ec38 _PyEval_EvalFrameDefault + 7384
	47  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	48  python3.10                          0x00000001009744a4 method_vectorcall + 124
	49  python3.10                          0x0000000100a850e8 _PyEval_EvalFrameDefault + 33160
	50  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	51  python3.10                          0x00000001009744a4 method_vectorcall + 124
	52  python3.10                          0x0000000100a850e8 _PyEval_EvalFrameDefault + 33160
	53  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	54  python3.10                          0x00000001009744a4 method_vectorcall + 124
	55  python3.10                          0x0000000100a850e8 _PyEval_EvalFrameDefault + 33160
	56  python3.10                          0x000000010096fddc _PyFunction_Vectorcall + 548
	57  python3.10                          0x00000001009fbb9c slot_tp_init + 540
	58  python3.10                          0x00000001009f10b8 type_call + 284
[... cut ...]
lang = env.get_ep_meta()["lang"]
print("Instruction:", lang)

for i in range(500):
    action = np.random.randn(*env.action_spec[0].shape) * 0.1
    obs, reward, done, info = env.step(action)  # take action in the environment
    env.render()  # render on display
```

In your case, if you run the above code snippet with `mjpython` in an interactive session, and you set that arg to False vs True, what happens? For me when it's False I get the `NSInternalInconsistencyException`, and with True it works as expected

felixzheng02, 2025-03-11:
Set `complex_config = True` in robo_kitchen.py works for me.

snasiriany, 2025-03-18:
I figured out the issue. The default viewer was set to the old viewer `mujoco` instead of the new viewer `mjviewer`. I just fixed the issue with this commit: https://github.com/robocasa/robocasa/commit/9f14a76cde2b87c473cbbc5a87eb975b80c2cab6

All of the problems you've been seeing now make sense. Apologies for that! Should be fixed with the latest code.

closed 2025-03-18 by snasiriany