robocasa/robocasa #124: Number of states different from number of actions

shaoyifei96, 2025-02-23:
Hello I was trying to see if the simulator is deterministic and I found the assertion always fails because in the demo the number of actions and states are different. (At least for OpenSingleDoor and PnPCounterToCab)

Following are a few prints:

```
Playing back episode: demo_1
Warning: number of actions 217 is not the same as number of states 267

Playing back episode: demo_2
Warning: number of actions 300 is not the same as number of states 350

Playing back episode: demo_3
Warning: number of actions 186 is not the same as number of states 236

Playing back episode: demo_4
Warning: number of actions 244 is not the same as number of states 294

```

Could you kindly let me know what is the best way to cut off the states trajectory? Should I just truncate it at the end?

snasiriany, 2025-02-23:
Hi @shaoyifei96 can you let us know the specific dataset paths you are using? Are these human/mimicgen datasets we already provide out of the box, or ones you generated? It would be great if you can share as many details as possible.

shaoyifei96, 2025-02-23:
I was running robocasa/demos/demo_tasks.py and changing the toggle line 137 to True