robocasa/robocasa #142: IQL algorithm training error

hyfqphy, 2025-04-29:
when i use the iql algorithm to train, execute the commad: "python /home/users_home/wyc/Project/robocasa/robomimic/robomimic/scripts/train.py --config /home/users_home/wyc/tmp/autogen_configs/ril/iql/robocasa/im/04-29-None/04-29-25-16-09-00/json/seed_123_ds_human-50.json"，the following error occurred

But when I use the bc_transformer algorithm to train, there is no error

```
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/opt/anaconda3/envs/robocasa/lib/python3.10/site-packages/torch/utils/data/_utils/worker.py", line 349, in _worker_loop
    data = fetcher.fetch(index)  # type: ignore[possibly-undefined]
  File "/opt/anaconda3/envs/robocasa/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/opt/anaconda3/envs/robocasa/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/users_home/wyc/Project/robocasa/robomimic/robomimic/utils/dataset.py", line 1096, in __getitem__
    meta = self.datasets[ds_ind].__getitem__(ind_in_ds)
  File "/home/users_home/wyc/Project/robocasa/robomimic/robomimic/utils/dataset.py", line 482, in __getitem__
    output = self.get_item(index)
  File "/home/users_home/wyc/Project/robocasa/robomimic/robomimic/utils/dataset.py", line 539, in get_item
    meta["next_obs"] = self.get_obs_sequence_from_demo(
  File "/home/users_home/wyc/Project/robocasa/robomimic/robomimic/utils/dataset.py", line 649, in get_obs_sequence_from_demo
    obs, pad_mask = self.get_sequence_from_demo(
  File "/home/users_home/wyc/Project/robocasa/robomimic/robomimic/utils/dataset.py", line 625, in get_sequence_from_demo
    data = self.get_dataset_for_ep(demo_id, k)
  File "/home/users_home/wyc/Project/robocasa/robomimic/robomimic/utils/dataset.py", line 472, in get_dataset_for_ep
    ret = self.hdf5_file[hd5key]
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/opt/anaconda3/envs/robocasa/lib/python3.10/site-packages/h5py/_hl/group.py", line 360, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5o.pyx", line 257, in h5py.h5o.open
KeyError: 'Unable to synchronously open object (component not found)'
```

snasiriany, 2025-04-29:
In this line:
```
  File "/home/users_home/wyc/Project/robocasa/robomimic/robomimic/utils/dataset.py", line 472, in get_dataset_for_ep
    ret = self.hdf5_file[hd5key]
```
what is the `hd5key` that is being specifically accessed here? Can you place a print line and see?

hyfqphy, 2025-05-01:
the hd5key contain these:

`data/demo_4/obs/robot0_agentview_right_image
data/demo_16/obs/robot0_agentview_right_image
data/demo_23/obs/robot0_agentview_right_image
data/demo_45/obs/robot0_gripper_qpos
data/demo_45/next_obs/robot0_agentview_left_image
data/demo_4/obs/robot0_base_pos
data/demo_4/obs/robot0_base_quat
data/demo_4/obs/robot0_base_to_eef_pos
data/demo_4/obs/robot0_base_to_eef_quat
data/demo_4/obs/robot0_eye_in_hand_image
data/demo_47/actions
data/demo_47/obs/robot0_agentview_left_image
data/demo_23/obs/robot0_base_pos
data/demo_23/obs/robot0_base_quat
data/demo_16/obs/robot0_base_pos
data/demo_23/obs/robot0_base_to_eef_pos
data/demo_16/obs/robot0_base_quat
data/demo_23/obs/robot0_base_to_eef_quat
data/demo_16/obs/robot0_base_to_eef_pos
data/demo_23/obs/robot0_eye_in_hand_image
data/demo_16/obs/robot0_base_to_eef_quat
data/demo_16/obs/robot0_eye_in_hand_image
data/demo_47/obs/robot0_agentview_right_image
data/demo_4/obs/robot0_gripper_qpos
data/demo_4/next_obs/robot0_agentview_left_image
data/demo_47/obs/robot0_base_pos
data/demo_47/obs/robot0_base_quat
data/demo_47/obs/robot0_base_to_eef_pos
data/demo_47/obs/robot0_base_to_eef_quat
data/demo_47/obs/robot0_eye_in_hand_image
data/demo_42/obs/robot0_gripper_qpos
data/demo_16/obs/robot0_gripper_qpos
data/demo_16/next_obs/robot0_agentview_left_image
data/demo_42/next_obs/robot0_agentview_left_image
data/demo_23/obs/robot0_gripper_qpos
data/demo_23/next_obs/robot0_agentview_left_image
data/demo_54/actions
data/demo_54/obs/robot0_agentview_left_image
data/demo_42/actions
data/demo_42/obs/robot0_agentview_left_image
data/demo_54/obs/robot0_agentview_right_image
data/demo_47/obs/robot0_gripper_qpos
data/demo_47/next_obs/robot0_agentview_left_image
data/demo_42/obs/robot0_agentview_right_image
data/demo_54/obs/robot0_base_pos
data/demo_54/obs/robot0_base_quat
data/demo_54/obs/robot0_base_to_eef_pos
data/demo_54/obs/robot0_base_to_eef_quat
data/demo_54/obs/robot0_eye_in_hand_image
data/demo_42/obs/robot0_base_pos
data/demo_42/obs/robot0_base_quat
data/demo_42/obs/robot0_base_to_eef_pos
data/demo_42/obs/robot0_base_to_eef_quat
data/demo_42/obs/robot0_eye_in_hand_image
data/demo_54/obs/robot0_gripper_qpos
data/demo_54/next_obs/robot0_agentview_left_image
data/demo_42/obs/robot0_gripper_qpos
data/demo_42/next_obs/robot0_agentview_left_image`