robocasa/robocasa #151: importerror when try example for BC-Transformer policy

XizhuoLiu, 2025-06-27:
First of all, thank you for having so many resources!

After downloading the data, I tried to run robocasa for the first time, the example in policy learning:
python robomimic/scripts/config_gen/bc_xfmr_gen.py --name 11
The following error occurred:

Traceback (most recent call last):
File "/home/username/robocasa/mytest/11.py", line 1, in <module> from huggingface_hub import list_repo_tree
ImportError: cannot import name 'list_repo_tree' from 'huggingface_hub'(/home/username/anaconda3/envs/robocasa/lib/python3.10/site-packages/huggingface_hub/__init__.py)

When I upgraded 'huggingface_hub', it included 'list_repo_tree', and re-ran it and it appeared:
ImportError: cannot import name 'cached_download' from 'huggingface_hub' (/home/username/anaconda3/envs/robocasa/lib/python3.10/site-packages/huggingface_hub/__init__.py)

I didn't find a huggingface_hub version that includes both functions. I think it's my mistake. No one in the issue encountered this problem, so I would like to ask how I should solve this problem. Thank you very much.

XizhuoLiu, 2025-07-03:
I have fixed the problem, it's just a version conflict. Thanks!

sakshi79, 2025-07-31:
> I have fixed the problem, it's just a version conflict. Thanks!

How did you fix it? What version fixed it for you?